GANESH-ICMC/esp32-deauther #1: Ninja error

andrewostroumov, 2019-11-27:
Hey hey!
I try to build your nice project but have some issues and maybe you may help me.
So what I do. I build with esp-idf master and have success with build and flash but cant connect to AP.
Then I try to checkout to a45e9985344575a80acb1dc9c12e7bec4d8af401 and build again
I have this error
```
-- Build files have been written to: /Users/andrewostroumov/esp/esp32-deauther/build
Running ninja in directory /Users/andrewostroumov/esp/esp32-deauther/build
Executing "ninja all"...
ninja: error: '/Users/andrewostroumov/esp/esp-idf/components/esp_wifi/lib_esp32/libcoexist.a', needed by 'deauther.elf', missing and no known rule to make it
ninja failed with exit code 1
```

andrewostroumov, 2019-11-27:
I use MacOS so maybe there are some system issues. I have to update patch.sh file to work with MacOS

andrewostroumov, 2019-11-27:
I'm new to ESP32 so sorry if my problem is obvious

henriquenunez, 2020-09-13:
> I use MacOS so maybe there are some system issues. I have to update patch.sh file to work with MacOS

Did it work?

utawkin2me, 2020-11-06:
i am having the same ninja issue...what is the solution i am on windows 10...

henriquenunez, 2020-11-06:
> i am having the same ninja issue...what is the solution i am on windows 10...

Hello, please send me the commit of the ESP idf you're using, and the commit number of the project.

henriquenunez, 2020-11-06:
> Hey hey!
> I try to build your nice project but have some issues and maybe you may help me.
> So what I do. I build with esp-idf master and have success with build and flash but cant connect to AP.
> Then [...]

have you tried cleaning the binaries?

utawkin2me, 2020-11-06:
I am new to dev boards and aside from the projects that my arduino came with, this is the first real project i have tried...

How do i find the commit number...i ran the build command in the esp32 deauth folder and that created a build directory...is in in there?

Why doesn't this repository have a detailed tutorial explaining the installation process....why haven't you built a .bin file containing everything so we could just flash the esp32 wroom 32? Don't get me wrong I like the hands on project because i am learning so much from trying to correct this error but i would've liked a easy install version as well to test out and use to test my wireless network at home...

I appreciate your help with this process...i imagine you get allot of emails...

Sent from ProtonMail mobile

-------- Original Message --------

henriquenunez, 2020-11-07:
> I am new to dev boards and aside from the projects that my arduino came with, this is the first real project i have tried... How do i find the commit number...i ran the build command in the esp32 [...]

Some considerations:

1. Attach the error you're getting. Otherwise I'm just reading "it did not work", and I'm unable to know if the problem is mine or yours.

2. There's no way of generating a binary for now, since the mac address is currently hardcoded. Doing it is a WIP, and will require some sort of interface between the user and the ESP32.

See you soon.

andrewostroumov, 2020-12-02:
I have new error
```
/Users/andrewostroumov/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/andrewostroumov/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/crt0.o:(.literal+0x0): undefined reference to `main'
```

JerryCloud, 2021-03-15:
Is this problem solved?I have the same problem:
 I build with esp-idf V4.3,but error:

ninja: error: '/home/jerry/esp/esp-idf/components/esp_wifi/lib/esp32/libwapi.a', needed by 'blink.elf', missing and no known rule to make it