GANESH-ICMC/esp32-deauther #3: Does not work

Garfius, 2020-02-28:
I have installed the exact version of esp-idf and it doesn't work, all wifi's are stable.

Procedure i have followed:
`git clone https://github.com/espressif/esp-idf

cd esp-idf

git reset --hard a45e9985344575a80acb1dc9c12e7bec4d8af401

/usr/bin/python -m pip install --upgrade virtualenv==16.7.9
`
Remove --no-site-packages from(unavailable option nowadays):
/home/myUser/Downloads/esp32Deauth/esp-idf/tools/idf_tools.py

`./install.sh

. ./export.sh

idf.py build

./patch.sh build/deauther.elf

7+0 records in
7+0 records out
7 bytes copied, 0.0019303 s, 3.6 kB/s

idf.py -p /dev/ttyS22 -b 921600 flash
`

Esp32 Output:
`
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:7244
load:0x40078000,len:14176
load:0x40080400,len:4480
entry 0x400806f0
 [0;32mI (62) boot: Chip Revision: 1 [0m
 [0;32mI (63) boot_comm: chip revision: 1, min. bootloader chip revision: 0 [0m
 [0;32mI (39) boot: ESP-IDF v4.1-dev-815-ga45e99853-dirty 2nd stage bootloader [0m
 [0;32mI (39) boot: compile time 17:46:05 [0m
 [0;32mI (40) boot: Enabling RNG early entropy source... [0m
 [0;32mI (45) boot: SPI Speed      : 80MHz [0m
 [0;32mI (50) boot: SPI Mode       : DIO [0m
 [0;32mI (54) boot: SPI Flash Size : 2MB [0m
 [0;32mI (58) boot: Partition Table: [0m
 [0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length [0m
 [0;32mI (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000 [0m
 [0;32mI (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000 [0m
 [0;32mI (83) boot:  2 factory          factory app      00 00 00010000 00100000 [0m
 [0;32mI (91) boot: End of partition table [0m
 [0;32mI (95) boot_comm: chip revision: 1, min. application chip revision: 0 [0m
 [0;32mI (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x15bec ( 89068) map [0m
 [0;32mI (138) esp_image: segment 1: paddr=0x00025c14 vaddr=0x3ffb0000 size=0x0329c ( 12956) load [0m
 [0;32mI (143) esp_image: segment 2: paddr=0x00028eb8 vaddr=0x40080000 size=0x00400 (  1024) load [0m
 [0;32mI (145) esp_image: segment 3: paddr=0x000292c0 vaddr=0x40080400 size=0x06d58 ( 27992) load [0m
 [0;32mI (164) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x6a5cc (435660) map [0m
 [0;32mI (295) esp_image: segment 5: paddr=0x0009a5f4 vaddr=0x40087158 size=0x09848 ( 38984) load [0m
 [0;32mI (320) boot: Loaded app from partition at offset 0x10000 [0m
 [0;32mI (320) boot: Disabling RNG early entropy source... [0m
 [0;32mI (321) cpu_start: Pro cpu up. [0m
 [0;32mI (324) cpu_start: Application information: [0m
 [0;32mI (329) cpu_start: Project name:     deauther [0m
 [0;32mI (334) cpu_start: App version:      ecedece-dirty [0m
 [0;32mI (339) cpu_start: Compile time:     Feb 28 2020 17:45:13 [0m
 [0;32mI (346) cpu_start: ELF file SHA256:  50620df0c76ce1ca... [0m
 [0;32mI (352) cpu_start: ESP-IDF:          v4.1-dev-815-ga45e99853-dirty [0m
 [0;32mI (358) cpu_start: Starting app cpu, entry point is 0x40081260 [0m
 [0;32mI (0) cpu_start: App cpu up. [0m
 [0;32mI (369) heap_init: Initializing. RAM available for dynamic allocation: [0m
 [0;32mI (376) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM [0m
 [0;32mI (382) heap_init: At 3FFB9240 len 00026DC0 (155 KiB): DRAM [0m
 [0;32mI (388) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM [0m
 [0;32mI (394) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM [0m
 [0;32mI (401) heap_init: At 400909A0 len 0000F660 (61 KiB): IRAM [0m
 [0;32mI (407) cpu_start: Pro cpu start user code [0m
 [0;32mI (425) spi_flash: detected chip: generic [0m
 [0;32mI (425) spi_flash: flash io: dio [0m
 [0;33mW (425) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header. [0m
 [0;32mI (436) cpu_start: Starting scheduler on PRO CPU. [0m
 [0;32mI (0) cpu_start: Starting scheduler on APP CPU. [0m
I (521) wifi: wifi driver task: 3ffc0b74, prio:23, stack:3584, core=0
 [0;32mI (521) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE [0m
 [0;32mI (521) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE [0m
I (551) wifi: wifi firmware version: af6bf13
I (551) wifi: config NVS flash: enabled
I (551) wifi: config nano formating: disabled
I (551) wifi: Init dynamic tx buffer num: 32
I (551) wifi: Init data frame dynamic rx buffer num: 32
I (561) wifi: Init management frame dynamic rx buffer num: 32
I (561) wifi: Init management short buffer num: 32
I (571) wifi: Init static rx buffer size: 1600
I (571) wifi: Init static rx buffer num: 10
I (571) wifi: Init dynamic rx buffer num: 32
 [0;32mI (1211) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0 [0m
I (1211) wifi: ic_enable_sniffer
I (1211) wifi: mode : softAP (4c:11:ae:c8:e5:75)
I (1221) wifi: Total power save buffer number: 16
I (1221) wifi: Init max length of beacon: 752/752
I (1221) wifi: Init max length of beacon: 752/752
I (1231) wifi: Set ps type: 0

Deauthing channel 1
Deauthing channel 2
I (2231) wifi: Total power save buffer number: 16
Deauthing channel 3
I (2231) wifi: Total power save buffer number: 16
Deauthing channel 4
I (2231) wifi: Total power save buffer number: 16
Deauthing channel 5
I (2241) wifi: Total power save buffer number: 16
Deauthing channel 6
I (2241) wifi: Total power save buffer number: 16
Deauthing channel 7
I (2251) wifi: Total power save buffer number: 16
Deauthing channel 8
`

SinanAkkoyun, 2020-03-29:
Did you try to enter the APs MAC in the main cpp?

zetanet, 2020-04-18:
Hi Gardius, I'm interested in this project, where can I download the Built v4.1-dev-763-ga45e99853? does it also work with the latest 4.1 dev on the site at https://github.com/espressif/esp-idf/releases/tag/v4.1-dev?

henriquenunez, 2020-09-13:
> I have installed the exact version of esp-idf and it doesn't work, all wifi's are stable.
>
> Procedure i have followed:
> `git clone https://github.com/espressif/esp-idf
>
> cd esp-idf
>
> git reset --hard [...]

Did you manage it to work by now?

henriquenunez, 2020-09-13:
> Hi Garfius, I'm interested in this project, where can I download the Built v4.1-dev-763-ga45e99853? does it also work with the latest 4.1 dev on the site at https://github.com/espressif/esp-idf/releases/tag/v4.1-dev?

Yes, it works with idf 4.1

closed 2020-11-07 by henriquenunez