GANESH-ICMC/esp32-deauther #11: Install Help

tonysoprano187, 2021-01-13:
Hey, anyone got any advice in installing this on a m5stack? Ideally I would like to install this on the lovyan launcher

many thanks!

EParisot, 2021-01-14:
Hi I absolutely know nothing about lovyan launcher but I struggled a bit to install as I'm on Windows10...
So I used WSL2 to compile (had to "idf.py build": it will fail but build some necessary files in the process, and then "make") after I installed esp-idf on wsl2 (git clone esp-idf, then "./install.sh" and ". ./export.sh").
And ultimately, since I discovered WSL2 does not support USB Serial communication, flashed it from the esptool.py from Windows directly...:
`#!/bin/sh
cd build
esptool.py -p COM3  -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 deauther.bin`

bootdsc, 2021-01-14:
how do I get it of this mail chain? please stop sending these to me.

EParisot, 2021-01-14:
That's fun your message felt right left to the "Unsubscribe" button ;)

EParisot, 2021-01-14:
EDIT: found out another way with use of this tool: https://matevarga.github.io/esp32/m5stack/esp-idf/wsl2/2020/05/31/flashing-esp32-under-wsl2.html
It streams the windows COM port over tcp and let you connect to it from wsl... kinda simpler since you just start it from powershell and do the rest (make, patch and flash) from wsl... 
ps: Sorry if you are not on Windows, just wanted to be complete here....

bootdsc, 2021-01-15:
please stop emailing me, this has been going on long enough. yes I've tried
to unsubscribe.

EParisot, 2021-01-15:
Man, I'm sorry to say that but this is an open area and if you received a mail notification, it's because you are somehow subscribed to it... Just unsubscribe from here (again ?) and stop "watching" the repo from its main page..., Or just quit GitHub...

The world can't stop running because it bother you...

tonysoprano187, 2021-01-15:
@bootdsc haha how did you even get involved.....

@EParisot my only thing is I use a Mac Osx... I basically want to be able to add games on to a menu system so would need a .bin file  

Many thanks :)

tobozo, 2021-02-18:
@tonysoprano187 in order to get this on loadable from LovyanLauncher you need to implement OTA rollback in your deauther.bin while making sure you're using a compatible partition scheme

ghaythbataineh06, 2024-10-04:
can anyone send it as .bin file i cant flash it