GANESH-ICMC/esp32-deauther #23: failed to compile reason:

ArshamEbr, 2023-11-23:
[901/902] Linking CXX executable deauther.elf
FAILED: deauther.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\deauther.elf.rsp  -o deauther.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Espressif/frameworks/esp-idf-v4.1.4/components/esp_wifi/lib/esp32/libnet80211.a(ieee80211_output.o): in function `ieee80211_raw_frame_sanity_check':
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; esp-idf/main/libmain.a(main.cpp.obj):c:\users\arsham\desktop\esp32\esp32-deauther-master\build/../main/main.cpp:38: first defined here
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

LeoNiik, 2023-12-11:
thanks, it worked!

jakydibe, 2023-12-13:
add this line to you CMakeList inside main: target_link_libraries(${COMPONENT_LIB} -Wl,-zmuldefs)